cra-ros-pkg/robot_localization #729: robots_localization:Transform from gyro_link to base_footprint was unavailable for the time requested. Using latest instead.

miku54, 2022-02-22:
## Bug report
[ WARN] [1645532392.821254984]: Transform from base_footprint to map was unavailable for the time requested. Using latest instead.
**Required Info:**

- Operating System:
  - ubuntu18.04 ros-melodic
Hello , when I use the robots_localization function package to fuse odom, imu, and gps data, the navsat_transform_node node will always flash the following alarms: "Transform from base_footprint to map was unavailable for the time requested. Using latest instead." The frequency of my /odom and /imu topics are both 50hz (but I don't think it has anything to do with the topic's frequency), the frequency of /gps/fix is 100hz, and the running files and parameters are in this repository:https://github.com/miku54/robot_localization-gps

miku54, 2022-02-22:
How should this situation be resolved?
If you need more information, please contact me, thank you

miku54, 2022-02-22:
This is the graph of Tf_tree and nodes when I debug[image]
[image]

ayrton04, 2022-02-22:
Please ask questions on answers.ros.org or robotics.stackexchange.com. If you have already done so, can you please provide a link? Also, if you do ask a question on one of those sites, please provide your config and sample messages from your sensor inputs.

closed 2022-02-22 by ayrton04

miku54, 2022-02-23:
https://answers.ros.org/question/396538/robots_localizationtransform-from-gyro_link-to-base_footprint-was-unavailable-for-the-time-requested-using-latest-instead/

miku54, 2022-02-23:
Hi, thank you for your reply! I have raised a related question in ros.org, the link is: https://answers.ros.org/question/396538/robots_localizationtransform-from-gyro_link-to-base_footprint-was-unavailable -for-the-time-requested-using-latest-instead/
Since there are no credits, my config files are placed in this repository: https://github.com/miku54/robot_localization-gps
Examples of sensor inputs are placed in this repository:
https://github.com/miku54/robot_localization-gps/tree/rosbag
Do you need more information?