cra-ros-pkg/robot_localization #732: navsat_transform crashes when broadcast_utm_transform is false

afrixs, 2022-02-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 3.3.0-1focal.20220122.083819

#### Steps to reproduce issue
Run `navsat_transform_node`, setting `broadcast_utm_transform` to false or not setting it at all (or setting 'broadcast_cartesian_transform' instead).

#### Expected behavior
Node runs normally without publishing utm transform

#### Actual behavior
Node crashes with message: parameter 'broadcast_utm_transform' has already been declared

#### Additional information
I believe [this line](https://github.com/cra-ros-pkg/robot_localization/blob/ffbe96b2286c1455f56efefca13796682333a8e4/src/navsat_transform.cpp#L119) was intended to be
```
      this->declare_parameter("broadcast_cartesian_transform", broadcast_cartesian_transform_);
```
instead of
```
      this->declare_parameter("broadcast_utm_transform", broadcast_cartesian_transform_);
```

ayrton04, 2022-02-25:
Thanks. Any chance you'd care to submit a PR? :)

savvas12, 2023-03-17:
Any solution for ros-noetic ? The navsat_transform.cpp is different for this version

ayrton04, 2023-03-30:
Noetic doesn't do parameter declaration, so I don't believe it should crash. What are you seeing in noetic?