cra-ros-pkg/robot_localization #740: echo /odometry/filtered no data

HusamZain, 2022-03-23:
Thank you for your great work.
I'm trying to use this package, so I can use it in NAV2 project .
I'm using Ros2 galactic on jetson nano, and I wanted to fuse data from wheel encoder and IMU 
I manged to publish the [nav_msgs/Odometry](http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html) on the **/odom** topic 
and [sensor_msgs/Imu](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html) on the **/imu** topic, also all the needed transformations are sat up (odom to base_link ) and (imu to base_link) 
so, my data are ready for fusion.
when I run the **ekf_node** and check the published topics from the node using `ros2 topic echo /odometry/filtered` I dont get any data.
I checked the **ekf_node** using `ros2 node info ` and it subscrubing to both **/odom** and **/imu** topics as below 
[image]
 and I checked both topics and they are publishing the **nav_msgs/Odometry** and **sensor_msgs/imu**  and everything looks just fine , does anyone has an idea what could be the issue that there are no data published on the ekf_node topics such as /odometry/filtered ?
I used `ros2 topic echo /diagnostics ` and this what I got, not sure if it's useful.

ayrton04, 2022-03-23:
Thanks for the question. However, all questions should be asked on answers.ros.org or robotics.stackexchange.com. When you do post a question, please include your config (and not as an image) and a sample message from each sensor input.

closed 2022-03-23 by ayrton04

tsadarsh, 2023-01-04:
Hi, @ayrton04. I have the exact same issue that is faced by @HusamZain. 

I have asked the question in answers.ros.org. [Here](https://answers.ros.org/question/411025/what-should-be-published-in-clock-when-using-a-real-robot/) is the link to my question. I also made sure to include the config file in the question. Can you please help me with this issue?

p.s. @HusamZain Did you get this issue solved already?

tsadarsh, 2023-01-05:
I solved this issue. The problem was with my *tf* tree.