cra-ros-pkg/robot_localization #749: Example launch file for dual_ekf_navsat_example contains wrong remapping rule

splatter96, 2022-05-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries 
- Version or commit hash:
  - galactic versions in apt as of today

#### Steps to reproduce issue
Set the wait_for_datum paramter to false
Launch the provided dual_ekf_navsat_example.launch.py launch file with an appropriate simulation of sensor setup.

#### Expected behavior
All transformations, especially map-->odom and odom-->baseframe should be published with the correct covariance.

#### Actual behavior
All transformations get published, but the map-->odom covariance increases with the moved distance.
After issuing a call to the /datum service with an initial pose, the covariance behaves as expected.

#### Additional information
After looking though the code of the navsat_transform node it show, that it subsribes to the "imu" topic for measurements of the imu sensor and only start proper transformations after it receives information at least once from it.
The provided launch file includes a remapping from "imu/data" to "imu/data" this remapping should be from "imu" to "imu/data". After changing this I got the expected behaviour.

ayrton04, 2022-05-03:
Thanks for catching that. Any chance you could submit a PR?

ayrton04, 2023-06-26:
If you want to submit a PR, please do.

SaiSugunSegu, 2023-06-26:
As I see, there is no issue in the code.

Note: **when using the Robot Localization Repository - any launch file - users need to remap the topic name to their own topic name (based on sensor output)**

SaiSugunSegu, 2023-06-26:
I don't feel its a bug, we can close this issue. 
@ayrton04 , please check and the close the issue, if you feel its a bug, please let me know, I want to work on it.

SaiSugunSegu, 2023-07-11:
Hey @joeldushouyu , 

I got it, I will change it to take it from the launch file, rather hardcoded topic in Navsat Transform cpp.

I will look at other nodes as well to correct them. Thanks.

joeldushouyu, 2023-07-11:
> Hey @joeldushouyu ,
> 
> I got it, I will change it to take it from the param file, rather hardcoded topic in Navsat Transform cpp.
> 
> I will look at other nodes as well to correct them. Thanks.

I think it will be better to just change how the topic got remapped in the example launch file. At least this is what I did.

SaiSugunSegu, 2023-07-12:
yeah, I will do the same.

joeldushouyu, 2023-07-17:
@SaiSugunSegu  
@splatter96  
Do you guys still plan to submit a PR for this fix?
I can do it if none of you guys are available

elgarbe, 2023-12-30:
please, someone file a PR for it, I was having the same issue and after a couple of hour I found this issue.

joeldushouyu, 2023-12-30:
@elgarbe  I have created a pending PR https://github.com/cra-ros-pkg/robot_localization/pull/857

elgarbe, 2023-12-30:
> @elgarbe I have created a pending PR #857

Thank!!! Hope this will be merge soon