cra-ros-pkg/robot_localization #764: Navsat_transform ekf Odometry/gps jumping to random values

Marcelomm103, 2022-07-11:
Hey you all!

I'm having troubles utilizing the navsat_transform node. My values of Odometry/gps are jumping from values too far from each other.

Expected behaviour:

The navsat transform the gps coordinates to xyz odometry values.

Actual behaviour:

The odometry/gps are too far from each other even if my robot is still, they are as follows after i set /datum:

header: stamp: sec: 9 nanosec: 500000000 frame_id: map child_frame_id: '' pose: pose: position: x: -14284.242740421556 y: 40063.96676235087 z: 0.0 orientation: x: 0.0 y: 0.0 z: 0.0 w: 1.0

header: stamp: sec: 9 nanosec: 400000000 frame_id: map child_frame_id: '' pose: pose: position: x: 51796.78036309546 y: -73793.62852960825 z: 0.0 orientation: x: 0.0 y: 0.0 z: 0.0 w: 1.0

header: stamp: sec: 9 nanosec: 300000000 frame_id: map child_frame_id: '' pose: pose: position: x: -109302.91985165933 y: -54294.5602690829 z: 0.0 orientation: x: 0.0 y: 0.0 z: 0.0 w: 1.0

header: stamp: sec: 9 nanosec: 200000000 frame_id: map child_frame_id: '' pose: pose: position: x: 55014.76906251442 y: 13793.76866940409 z: 0.0 orientation: x: 0.0 y: 0.0 z: 0.0 w: 1.0

I dont think this is the normal behaviour, is this a bug?

ayrton04, 2022-07-12:
Thanks for the question. However, all questions should be directed to answers.ros.org or robotics.stackexchange.com. When you ask your question, please include your full configuration file and a sample message from each sensor input.

closed 2022-07-12 by ayrton04

Marcelomm103, 2022-07-14:
Actualy i have managed to fix the bug on the jump values but my costmap is moving its obstacles detected with the robot as it goes: https://answers.ros.org/question/403748/obstacles-moving-together-with-local-costmap/

Marcelomm103, 2022-07-14:
Actualy i managed to correct the jumping issue, but my tf map -> odom is moving together with the robot causing the obstacles in the localcostmap to move with the robot as well. https://answers.ros.org/question/403748/map-odom-tf-moving-causing-obstacles-in-localcostmap-to-move-with-robot/