cra-ros-pkg/robot_localization #766: plain cmake error

YXGuan, 2022-07-18:
Hi there,

I got a Cmake error when trying to build the robot_localization package on Noetic.

Your help would be much appreciated.

Below are the output in the terminal

```
<pre><font color="#4E9A06"><b>yuxiang@yuxiang</b></font>:<font color="#3465A4"><b>~/catkin_ws</b></font>$ catkin_make
Base path: /home/yuxiang/catkin_ws
Source space: /home/yuxiang/catkin_ws/src
Build space: /home/yuxiang/catkin_ws/build
Devel space: /home/yuxiang/catkin_ws/devel
Install space: /home/yuxiang/catkin_ws/install
<font color="#3465A4">####</font>
<font color="#3465A4">#### Running command: </font><font color="#3465A4"><b>&quot;cmake /home/yuxiang/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/yuxiang/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yuxiang/catkin_ws/install -G Unix Makefiles&quot;</b></font><font color="#3465A4"> in </font><font color="#3465A4"><b>&quot;/home/yuxiang/catkin_ws/build&quot;</b></font>
<font color="#3465A4">####</font>
-- Using CATKIN_DEVEL_PREFIX: /home/yuxiang/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/yuxiang/catkin_ws/devel;/home/yuxiang/ydlidar_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/yuxiang/catkin_ws/devel;/home/yuxiang/ydlidar_ws/devel;/opt/ros/noetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: gps_waypoint_nav;robot_localization;tf2;tf2_eigen;tf2_geometry_msgs;tf2_msgs;tf2_py;tf2_ros
-- <font color="#75507B">~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</font>
-- <font color="#75507B">~~ </font> traversing 7 packages in topological order:
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2_msgs</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2_eigen</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2_py</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2_ros</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>tf2_geometry_msgs</b></font>
-- <font color="#75507B">~~ </font> - <font color="#3465A4"><b>robot_localization </b></font>(<font color="#CC0000">plain cmake</font>)
-- <font color="#75507B">~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</font>
<font color="#CC0000"><b>CMake Error</b></font> at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  &apos;catkin_make_isolated&apos; command instead.
<font color="#06989A"><u style="text-decoration-style:single">Call Stack (most recent call first):</u></font>
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also &quot;/home/yuxiang/catkin_ws/build/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/yuxiang/catkin_ws/build/CMakeFiles/CMakeError.log&quot;.
<font color="#CC0000">Invoking </font><font color="#CC0000"><b>&quot;cmake&quot;</b></font><font color="#CC0000"> failed</font>
<font color="#4E9A06"><b>yuxiang@yuxiang</b></font>:<font color="#3465A4"><b>~/catkin_ws</b></font>$ catkin_make_isolated
Base path: /home/yuxiang/catkin_ws
Source space: /home/yuxiang/catkin_ws/src
Build space: /home/yuxiang/catkin_ws/build_isolated
Devel space: /home/yuxiang/catkin_ws/devel_isolated
Install space: /home/yuxiang/catkin_ws/install_isolated
<font color="#75507B">~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</font>
<font color="#75507B">~~</font>  traversing 94 packages in topological order:
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>beginner_tutorials</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>costmap_cspace_msgs</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>ddynamic_reconfigure</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>delta_description</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>delta_viz_server</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>exploration</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>exploration_msgs</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>followbot</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>geometry2</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>learning_tf</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>map_organizer_msgs</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>my_camera</b></font>
[... 64 lines cut ...]
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>depthai_examples</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>voxel_grid</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>costmap_2d</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>explore_lite</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>nav_core</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>base_local_planner</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>carrot_planner</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>clear_costmap_recovery</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>dwa_local_planner</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>move_slow_and_clear</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>navfn</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>global_planner</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>polygon_layer</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>exploration_server</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>frontier_exploration</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>rotate_recovery</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>move_base</b></font>
<font color="#75507B">~~</font>  - <font color="#3465A4"><b>rtabmap_ros</b></font>
<font color="#75507B">~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</font>
Error: Packages with unknown build types exist
Can not build workspace with packages of unknown build_type
<font color="#4E9A06"><b>yuxiang@yuxiang</b></font>:<font color="#3465A4"><b>~/catkin_ws</b></font>$ 

</pre>
```

ayrton04, 2022-07-18:
This is not an issue with robot_localization. Please ask your question on answers.ros.org.

closed 2022-07-18 by ayrton04

divi2504, 2022-12-10:
@YXGuan I am having same error, can you suggest how you resolved it, if you did?

larryferguson, 2023-04-24:
ayronton04 thank you so much for being completely unhelpful to anyone. I get the same error building the package after Ive just installed it from source.

ayrton04, 2023-04-24:
@larryferguson the package builds on the ROS build farm. I'd say if anyone is being unhelpful, it's people posting troll comments instead of attempting to contribute fixes.

I've given many hundreds of hours of my own time to the development, maintenance, and support of this package. If you're unhappy with the level of free support provided, I recommend you write your own.

larryferguson, 2023-04-24:
> @larryferguson the package builds on the ROS build farm. I'd say if anyone is being unhelpful, it's people posting troll comments instead of attempting to contribute fixes.
>
> I've given many hundreds [...]

Sorry for my rudeness. I took out frustration towards you instead of the actual issue. I still disagree, this is the correct place to ask a question since it is directly about this package and that even when installed from source it will not catkin_make. I find that issue important to fix for the noetic version of this package.

ayrton04, 2023-04-25:
The reason I direct people to answers.ros.org is that if others are having the same issue, the entire community benefits from the answer.

Can you cd to the `robot_localization` directory on your machine and run `git branch`? The default branch for `rl` is `ros2`, so if you don't explicity checkout `noetic-devel`, it will see it as an ament package, which catkin doesn't understand.

ayrton04, 2023-04-26:
@larryferguson @gupta-divi @YXGuan can you please confirm the branch of `r_l` that you were trying to build? `cd` to the `r_l` directory and run `git branch`.

cst0, 2023-10-18:
My apologies for commenting on a closed issue, but I found this when I encountered this issue myself so I'm following up with a fix for others:

This error comes up for ROS 1 users if they (like me) accidentally leave their branch set to a ROS 2 branch (like the default one). ROS 2 uses `ament` and so `catkin` is nowhere to be found, hence the confusion for `catkin_make` (or `catkin build`).

The fix is to make sure you're on a ROS 1 based branch (I used `noetic-devel`) and then everything works fine.