cra-ros-pkg/robot_localization #771: rotational offset between odometry/gps and odometry/filtered

oskarmue, 2022-08-19:
Hi, 
for the odometry/filtered I use the ekf_node in order to fuse a lidar_odometry source and imu data. Then I use odometry/filtered together with the imu/data for the navsat_transform node. It all seems to work fine, except that the Position from odometry/gps seems to have a roational offset from the Position calculated from lidar and imu (see image below, white is odometry/gps and red is odometry/filtered). Are there certain parameters in order to correct that?
[image]

`

this is my param file

`
ekf_filter_node:
  ros__parameters:
    frequency: 30.0
    sensor_timeout: 2.0
    two_d_mode: false
    transform_time_offset: 0.0
    transform_timeout: 0.0
    print_diagnostics: true
    debug: false

    map_frame: map
    odom_frame: odom
    base_link_frame: base_link
    world_frame: odom

    odom0: /lidarslam_odometry
    odom0_config: [true, true, true,
                  false, false, false,
                  false, false, false,
                  false, false, false,
                  false, false, false]
    odom0_queue_size: 10
    odom0_differential: false
    odom0_relative: true

    imu0: imu/data
    imu0_config: [false, false, false,
                  true,  true,  true,
                  false, false, false,
                  true,  true,  true,
                  true,  true,  true]

    imu0_nodelay: false
    imu0_differential: false
    imu0_relative: true
    imu0_queue_size: 5
    imu0_pose_rejection_threshold: 0.8                 # Note the difference in parameter names
    imu0_twist_rejection_threshold: 0.8                #
    imu0_linear_acceleration_rejection_threshold: 0.8  #

    imu0_remove_gravitational_acceleration: true

    use_control: false
navsat_node:
  ros__parameters:
    frequency: 30.0
    delay: 3.0
    magnetic_declination_radians: 0.058 #0.04256444  
    yaw_offset: 0.0 
    zero_altitude: false
    broadcast_cartesian_transform: false
    #broadcast_utm_transform_as_parent_frame: false
    publish_filtered_gps: false
    use_odometry_yaw: false
    wait_for_datum: false`

oskarmue, 2022-08-21:
Could the reason for that be a magnetic field, that is influencing the imu and therefore the direction of the east is not measured correctly?

ayrton04, 2022-08-22:
Thanks for the question. However, all questions should be directed to answers.ros.org or robotics.stackexchange.com.

closed 2022-08-22 by ayrton04