cra-ros-pkg/robot_localization #772: ROS2 Humble Debian Package: parameter 'broadcast_utm_transform' has already been declared

austin-InDro, 2022-08-22:
Hello,

There seems to be an issue with the current debian install of ros-humble-robot-localization
```
apt-cache policy ros-humble-robot-localization
ros-humble-robot-localization:
  Installed: 3.3.1-2jammy.20220808.154630
  Candidate: 3.3.1-2jammy.20220808.154630
  Version table:
 *** 3.3.1-2jammy.20220808.154630 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status

```
Issue:
```
ros2 run robot_localization navsat_transform_node
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
  what():  parameter 'broadcast_utm_transform' has already been declared
[ros2run]: Aborted
```

Possible Temporary fix
```
ros2 run robot_localization navsat_transform_node --ros-args -p broadcast_utm_transform:=true
[WARN] [1661179491.328314343] [navsat_transform_node]: Parameter 'broadcast_utm_transform' has been deprecated. Please use 'broadcast_cartesian_transform' instead.
^C[INFO] [1661179492.619588515] [rclcpp]: signal_handler(signum=2)

```

I've observed this error on ARM and on x86 systems. The issue was also mentioned here on the ros forms: https://answers.ros.org/question/378284/robot_localization-navsat_transform_node-runtime-error/

However, it still seems to be prominent

ayrton04, 2022-08-22:
https://github.com/cra-ros-pkg/robot_localization/issues/732
https://github.com/cra-ros-pkg/robot_localization/pull/733

Just needs to be ported and bloomed.

austin-InDro, 2022-08-22:
Oh nice thanks!

ijnek, 2022-11-23:
@ayrton04 May I request for a bloom in Humble and Rolling please? I believe the changes have already been ported.

ayrton04, 2022-12-14:
I believe Steve did this.

closed 2022-12-14 by ayrton04