cra-ros-pkg/robot_localization #779: Parametrization of Rotation

Jad-Tawil, 2022-11-08:
After inspecting the code, it seems like the parameterization of rotation is done in roll / pitch / yaw angles. The innovation is normalized to avoid discontinuous jumps around pi/-pi, and after the state is updated, the angles are warped.
I have read elsewhere that using roll / pitch / yaw is unadvisable because they are discontinuous and singular. Using quaternions is a better option. 
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6339217/

Would you care to comment on this ?

ayrton04, 2022-11-08:
Yes, well aware that quaternions are preferable for numerous reasons. But this package is really in a "maintain" state at this point, with our focus moving to [fuse](https://github.com/locusrobotics/fuse). So I don't see it changing.

Jad-Tawil, 2022-11-08:
Thank you Tom, I am looking into fuse atm. I see it is using graph optimization instead of kalman filter. In graph optimization, is the parameterization of rotation i mentioned at play?

ayrton04, 2022-11-09:
The kinematic model is independent of the underlying sensor fusion algorithm. Kalman filters can use quaternions and factor graphs can use Euler angles.

Unlike this package, which assumes a single kinematic model, `fuse` is built from the ground-up to allow any kind of kinematic model (or sensor model). For example, the package has constraints for 3D rotations represented as quaternions or Euler angles:

https://github.com/locusrobotics/fuse/blob/devel/fuse_constraints/src/absolute_orientation_3d_stamped_constraint.cpp
https://github.com/locusrobotics/fuse/blob/devel/fuse_constraints/src/absolute_orientation_3d_stamped_euler_constraint.cpp

It's up to the user to work with (or create) the plugins that are best suited to their problem.

Jad-Tawil, 2022-11-09:
That makes sense! Looks like a great package. Well done!

ayrton04, 2022-11-09:
Credit is entirely to my colleague @svwilliams!