cra-ros-pkg/robot_localization #780: Square-root UKF for better numerical stability

fabianhirmann, 2022-11-17:
In our robotics group we noticed that sometimes the UKF seems to be unstable without any clearly noticeable reason. Unfortunately we don't have (yet) any rosbag at hand when the situation occurred. 
However, I just remembered that there exists also a square-root form of the UKF which should have better numerical stability [[1]](https://www.researchgate.net/publication/3908304_The_Square-Root_Unscented_Kalman_Filter_for_State_and_Parameter-Estimation) and as far as I noticed in the source code, this variant is not used here.

What do you think about that? Would the square-root UKF be an option to increase general numerical stability or do you think it might not make much difference? Or is this (or ideas from it) already implemented and I didn't see it?

If this feature already exists, feel free to close this feature request.

Thanks,
Fabian

ayrton04, 2022-11-17:
Oh, I should also note that this package is no longer adding new features. I would encourage users to switch to [fuse](https://github.com/locusrobotics/fuse).

fabianhirmann, 2022-11-18:
Thanks for your quick reply! It is interesting and good to hear that you recommend switching to fuse because we are currently actually in the process of that.

In any case, I spent now some time finding a situation where a strange behavior occurs. Very quickly summarized the pose there suddenly jumps from y = ~0 m to y = ~ -9m and the other data seems fine. We then "resolved" the issue by switching to the EKF which showed no problems like that afterwards when testing it directly at the robot.

Here's the configuration (with the UKF). It is important to note that /initial_current_pose just gets published once at restart and the NDT at /ndt_pose_with_covariance gets feed back the UKF pose and uses this as current pose estimation:
ukf.launch
```
<?xml version="1.0"?>
<launch>
  <node name="ukf" pkg="robot_localization" type="ukf_localization_node" clear_params="true" respawn="true">

    <param name="predict_to_current_time" value="true"/>
    <param name="dynamic_process_noise_covariance" value="true"/>
    <param name="smooth_lagged_data" value="true"/>
    <param name="history_length" value="2.0"/>

    <param name="frequency" value="50"/>
    <param name="sensor_timeout" value="0.2"/>
    <param name="transform_time_offset" value="0.07"/> 
    <param name="publish_tf" value="true"/>

    <param name="two_d_mode" value="true"/>

    <param name="map_frame" value="map"/>
    <param name="base_link_frame" value="base_link"/>
    <param name="world_frame" value="map"/>

    <param name="use_control" value="true"/>
    <rosparam param="control_config">[true,  false, false,
                                     false, false, true]</rosparam>

    <param name="reset_on_time_jump" value="true"/>

    <param name="print_diagnostics" value="true"/>

    <param name="pose0" value="/initial_current_pose"/>
    <rosparam param="pose0_config">[true,  true,  false,
                                    false, false, true, 
                                    false, false, false, 
                                    false, false, false,
                                    false, false, false]</rosparam>
    <param name="pose0_differential" value="false"/>
    <param name="pose0_relative" value="false"/>

    <param name="pose1" value="/ndt_pose_with_covariance"/>
    <rosparam param="pose1_config">[true,  true,  true,
                                    true,  true,  true, 
                                    false, false, false, 
                                    false, false, false,
                                    false, false, false]</rosparam>
    <param name="pose1_differential" value="false"/>
    <param name="pose1_relative" value="false"/>

    <param name="imu0" value="/imu/data"/>
    <rosparam param="imu0_config">[false, false, false,
                                   false, false, false,
                                   false, false, false,
                                   true, true, true,
                                   true, true, true]</rosparam>
    <param name="imu0_differential" value="true"/>
    <param name="imu0_remove_gravitational_acceleration" value="true"/>
<!-- if can is not working this is bullshit for localization -->
    <param name="odom0" value="/odom"/>
    <rosparam param="odom0_config">[false, false,  false, 
                                    false, false, false, 
                                    true, true,  true, 
                                    true, true, true,
                                    false, false, false]</rosparam>
    <param name="odom0_differential" value="false"/>
    <param name="odom0_relative" value="false"/>
[... 774 lines cut ...]
  seq: 2378
  stamp: 
    secs: 1668179164
    nsecs: 227426000
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 0.282901406288
      y: -9.90779304504
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.00486808678127
      w: 0.999988150795
  covariance: [0.002, 8.120300276689795e-05, 0.0, 0.0, 0.0, 0.0, 8.120300276687302e-05, 0.002, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

From the rosbags that I found unfortunately there was no messages on topics `/imu/data` and `/initial_current_pose` recorded. 

We are using ROS melodic and robot_localization was installed using rosdep (the robot is currently not here so I can't tell which exact version was used).

If there is anything else I can provide, please let me know. Already now thanks for your great help!