cra-ros-pkg/robot_localization #790: Unknown CMake command "rosidl_get_typesupport_target".

SubMishMar, 2023-01-12:
I get the following error on doing `colcon build`

[image]

SubMishMar, 2023-01-12:
I switched branch to `foxy-devel` and now I get this:

```
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/set_datum__functions.h:19,
                 from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/set_datum__functions.c:4:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/set_datum__struct.h:22:10: fatal error: geographic_msgs/msg/detail/geo_pose__struct.h: No such file or directory
   22 | #include "geographic_msgs/msg/detail/geo_pose__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/from_ll__functions.h:19,
                 from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/from_ll__functions.c:4:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/from_ll__struct.h:22:10: fatal error: geographic_msgs/msg/detail/geo_point__struct.h: No such file or directory
   22 | #include "geographic_msgs/msg/detail/geo_point__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/to_ll__functions.h:19,
                 from /home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/to_ll__functions.c:4:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_c/robot_localization/srv/detail/to_ll__struct.h:45:10: fatal error: geographic_msgs/msg/detail/geo_point__struct.h: No such file or directory
   45 | #include "geographic_msgs/msg/detail/geo_point__struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/robot_localization__rosidl_generator_c.dir/build.make:238: CMakeFiles/robot_localization__rosidl_generator_c.dir/rosidl_generator_c/robot_localization/srv/detail/from_ll__functions.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/robot_localization__rosidl_generator_c.dir/build.make:264: CMakeFiles/robot_localization__rosidl_generator_c.dir/rosidl_generator_c/robot_localization/srv/detail/set_datum__functions.c.o] Error 1
make[2]: *** [CMakeFiles/robot_localization__rosidl_generator_c.dir/build.make:303: CMakeFiles/robot_localization__rosidl_generator_c.dir/rosidl_generator_c/robot_localization/srv/detail/to_ll__functions.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:927: CMakeFiles/robot_localization__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_typesupport_cpp/robot_localization/srv/set_datum__type_support.cpp:7:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_cpp/robot_localization/srv/detail/set_datum__struct.hpp:19:10: fatal error: geographic_msgs/msg/detail/geo_pose__struct.hpp: No such file or directory
   19 | #include "geographic_msgs/msg/detail/geo_pose__struct.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_typesupport_cpp/robot_localization/srv/from_ll__type_support.cpp:7:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_cpp/robot_localization/srv/detail/from_ll__struct.hpp:19:10: fatal error: geographic_msgs/msg/detail/geo_point__struct.hpp: No such file or directory
   19 | #include "geographic_msgs/msg/detail/geo_point__struct.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_typesupport_cpp/robot_localization/srv/to_ll__type_support.cpp:7:
/home/smishr30/ros2_ws/build/robot_localization/rosidl_generator_cpp/robot_localization/srv/detail/to_ll__struct.hpp:130:10: fatal error: geographic_msgs/msg/detail/geo_point__struct.hpp: No such file or directory
  130 | #include "geographic_msgs/msg/detail/geo_point__struct.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/build.make:185: CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/robot_localization/srv/set_datum__type_support.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/build.make:159: CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/robot_localization/srv/from_ll__type_support.cpp.o] Error 1
make[2]: *** [CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/build.make:224: CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/rosidl_typesupport_cpp/robot_localization/srv/to_ll__type_support.cpp.o] Error 1
In file included from /home/smishr30/ros2_ws/build/robot_localization/rosidl_typesupport_introspection_cpp/robot_localization/srv/detail/set_datum__type_support.cpp:12:
[... cut ...]
make[2]: *** [CMakeFiles/robot_localization__rosidl_typesupport_fastrtps_cpp.dir/build.make:179: CMakeFiles/robot_localization__rosidl_typesupport_fastrtps_cpp.dir/rosidl_typesupport_fastrtps_cpp/robot_localization/srv/detail/dds_fastrtps/from_ll__type_support.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1397: CMakeFiles/robot_localization__rosidl_typesupport_cpp.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1370: CMakeFiles/robot_localization__rosidl_typesupport_introspection_cpp.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1316: CMakeFiles/robot_localization__rosidl_typesupport_fastrtps_cpp.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

ayrton04, 2023-01-15:
This builds on the build farm, so your issues are likely to be related to youre setup. Please make sure you using the right base ROS installation.

closed 2023-01-15 by ayrton04