cra-ros-pkg/robot_localization #791: Localizing multiple axles for an articulated vehicle

alexanderjyuen, 2023-01-25:
Has there any work been done for localizing multiple points with multiple sensors for a vehicle? To elaborate, imagine a truck pulling a trailer with a sensor suite (gps and imu) for the truck's rear axle (classic base link) and the trailer's rear axle (the trailer's base link).

In this scenario I would imagine two EKFs running, one for the truck and one for the trailer, and your controller and planner will processes it in whatever way it sees fit. Is this something that has been/or is being worked on? Is this something can already be done by configuring robot localization properly?

Any advice would be greatly appreciated

SteveMacenski, 2023-01-25:
This is not the right forum for this question

closed 2023-01-25 by SteveMacenski

ayrton04, 2023-01-25:
Yeah, I'd recommend using answers.ros.org.