cra-ros-pkg/robot_localization #799: Create a separated package for services and messages

Myzhar, 2023-02-28:
It would be useful if the declaration of the custom messages and services was in a separate package from the `robot_localization` source code.
In this way would be possible to subscribe to topics on machines where the full `robot_localization` package is not required and so make a cleaner robot installation.

ayrton04, 2023-03-09:
I feel like that would only be useful if those messages were added to a core ROS package. The `r_l` package isn't that large, so having it on all the machines that need to work with the package's messages doesn't seem a huge burden.

Myzhar, 2023-03-29:
@ayrton04 indeed `robot_localization` is relatively small, but it brings other dependencies with it.
When you want to create a small Docker image this is not good.

ayrton04, 2023-04-06:
While I appreciate the ask, I am unlikely to make any changes like this, as my aim is to get people to move to [fuse](https://github.com/locusrobotics/fuse).

closed 2023-04-07 by SteveMacenski