cra-ros-pkg/robot_localization #801: robot_localization not publishing

an99990, 2023-04-02:
HI,
i am trying to fuse my sensors and rtabmap visual odom topics together. However, robot_localization doesnt output anything. The diagnostic topic also doesnt tells me anything.
Here is a lonk to my rosbag 

launching rtabmap : 
`rosrun rtabmap_ros rgbd_odometry rgb/image:=/stereo_inertial_publisher/left/image_rect depth/image:=/stereo_inertial_publisher/stereo/depth rgb/camera_info:=/stereo_inertial_publisher/left/camera_info _frame_id:=oak-d_frame`

launching robot_localization:
`roslaunch robot_localization ekf_local.launch viz_veh:=false rviz:=false`

Here is my yaml
```
frequency: 30
two_d_mode: true
transform_time_offset: 0.0
transform_timeout: 0.02
print_diagnostics: true
debug: false
debug_out_file: /home/catkin_ws/src/debug.txt
publish_tf: false
publish_acceleration: false
map_frame: map                   # Defaults to "map" if unspecified
odom_frame: odom            # Defaults to "odom" if unspecified
base_link_frame: r_robot  # Defaults to "base_link" if unspecified
world_frame: odom           # Defaults to the value of odom_frame if unspecified

odom0: /odometry/filtered
odom0_config: [false, false, false,   # x y z
               true, false, false,   # roll pitch yaw
               false,  true,  false,   # vx vy vz
               false, false, true,    # vroll vpitch vyaw
               false, false, false]   # ax ay az
odom0_queue_size: 10
odom0_nodelay: false
odom0_differential: false
odom0_relative: false

odom1: /odom_rgbd_image
odom1_config: [false, false, false,   # x y z
               false, false, false,   # roll pitch yaw
               true,  true,  false,   # vx vy vz
               false, false, true,    # vroll vpitch vyaw
               false, false, false]   # ax ay az
               
odom1_queue_size: 10
odom1_nodelay: false
odom1_differential: false
odom1_relative: false

use_control: false
stamped_control: false
control_timeout: 0.2
control_config: [true, false, false, false, false, true]
process_noise_covariance: [0.05, 0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0.05, 0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0.06, 0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0.03, 0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0.03, 0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0.06, 0,     0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0.025, 0,     0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0.025, 0,    0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0.04, 0,    0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0.01, 0,    0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0.01, 0,    0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0.02, 0,    0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0.01, 0,    0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0.01, 0,
                           0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0.015]

# [ADVANCED] This represents the initial value for the state estimate error covariance matrix. Setting a diagonal
# value (variance) to a large value will result in rapid convergence for initial measurements of the variable in
# question. Users should take care not to use large values for variables that will not be measured directly. The values
# are ordered as x, y, z, roll, pitch, yaw, vx, vy, vz, vroll, vpitch, vyaw, ax, ay, az. Defaults to the matrix below
#if unspecified.
initial_estimate_covariance: [1e-9, 0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    1e-9, 0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    1e-9, 0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    1e-9, 0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    1e-9, 0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    1.0,  0,    0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    1e-9, 0,    0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    1e-9, 0,    0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    1e-9, 0,     0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    1e-9,  0,     0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     1e-9,  0,     0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     1e-9,  0,    0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     1e-9, 0,    0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    1e-9, 0,
                              0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    1e-9]
```

SteveMacenski, 2023-04-02:
Ask on ROS Answers

closed 2023-04-02 by SteveMacenski

an99990, 2023-04-02:
i never posted on ros so im not sure if itll get answered : https://answers.ros.org/question/413984/robot_localization-not-publishing/

ayrton04, 2023-04-21:
I will try to get to it when I have time, thanks.