cra-ros-pkg/robot_localization #802: Porting version 3.3.2 to Ubuntu 22.04 Jammy package list

mzuzic-maidea, 2023-04-03:
Hi,
There was an issue between 3.3.1 and 3.3.2 with 'broadcast_utm_transform' parameter being declared 2 times (typo) and I have following issue. Since I am using RPI4, for some unknown reason (memory leak, cpu issue not sure) I am not able to build robot-localization package inside ros:humble docker image. 

Ideally, It would be better if you could somehow update ubuntu package list to bump robot_localization from 3.3.1 version to 3.3.2 which is latest stable.

If we look at the Ubuntu Jammy package list, we see 3.3.1 version instead latest stable release 3.3.2.

http://packages.ros.org/ros2/ubuntu/dists/jammy/main/binary-arm64/Packages

Package: ros-humble-robot-localization
Version: 3.3.1-2jammy.20230327.050731
Architecture: arm64
Maintainer: Tom Moore <ayrton04@gmail.com>
Installed-Size: 34613
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), libgeographic19 (>= 1.40), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), libyaml-cpp0.7 (>= 0.7.0), ros-humble-fastcdr, libboost-dev, libeigen3-dev, libgeographic-dev, ros-humble-diagnostic-msgs, ros-humble-diagnostic-updater, ros-humble-geographic-msgs, ros-humble-geometry-msgs, ros-humble-message-filters, ros-humble-nav-msgs, ros-humble-rclcpp, ros-humble-rmw-implementation, ros-humble-rosidl-default-runtime, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-tf2, ros-humble-tf2-eigen, ros-humble-tf2-geometry-msgs, ros-humble-tf2-ros, ros-humble-yaml-cpp-vendor, ros-humble-ros-workspace
Homepage: http://ros.org/wiki/robot_localization
Priority: optional
Section: misc
Filename: pool/main/r/ros-humble-robot-localization/ros-humble-robot-localization_3.3.1-2jammy.20230327.050731_arm64.deb
Size: 5716802
SHA256: caf902fdb352469e5238ff3f056b9df6401e59ae05c8d3be5934a044978eb8da
SHA1: ab3aba8893b41c650282746db788ec6fc670c357
MD5sum: d5fb2c9aa95d2170e69fef7c1388ae82
Description: Provides nonlinear state estimation through sensor fusion of an abritrary number of sensors.

There is ros-rolling-robot_localization available but it is version 3.5 and it requires lot of other package dependencies being updated.

Is there a way to port version 3.3.2 to Ubuntu Jammy package list?

SteveMacenski, 2023-04-05:
There are multiple distributions on Jammy. I assume you mean Humble given your discussion on docker. Tom did 3.3.1/3.3.2 so I'll let him chime in but assuming there's not something subtle or API breaking changes that seems doable - I can run bloom over it if he's OK with it.

mzuzic-maidea, 2023-04-06:
Hey, thank you for your response. Hope it will get resolved.

I did manage to workaround the issue by setting broadcast_utm_transform to true so it does not enter into else condition which has the parameter declaration bug but still, would be nice to port it.

Yes, I was talking about humble edition, ros-humble-robot-localization

ayrton04, 2023-04-06:
Yeah, go for it @SteveMacenski, thanks.

SteveMacenski, 2023-04-14:
Oh it actually looks like I tried awhile ago @ayrton04 but I needed you to approve the PR https://github.com/cra-ros-pkg/robot_localization/pull/789. Since its my own PR, the settings you have setup block me from merging myself (maybe I should get elevated permissions at this point :laughing: )

ayrton04, 2023-04-14:
Merged. I've also elevated your permissions to "Maintain."

SteveMacenski, 2023-04-14:
@ayrton04 exact same issue in https://github.com/cra-ros-pkg/robot_localization/pull/806. I think you need to give maintainers a bypass on protected branches

ayrton04, 2023-04-17:
Try again?