cra-ros-pkg/robot_localization #808: fusing IMU and encoder not working for me

foxydevelby, 2023-05-03:
I just wanted to thank you for creating this package, however what could be the possible reasons for the failure of fusing IMU and wheel encoder data in my robot's localization system and how can I troubleshoot this issue?

Case: Robot is stuck in obstacle(not moving) but wheels are continuously rotating.

Expected Output:
base_link will not move and should be in place

Actual Output:
base_link moves continuously. (as seen)

[ekf.txt](https://github.com/cra-ros-pkg/robot_localization/files/11380360/ekf.txt)
[image]

ayrton04, 2023-05-03:
Please ask this question on answers.ros.org and then post the link here, and I'll get to it soon.

closed 2023-05-03 by ayrton04

foxydevelby, 2023-05-03:
> Please ask this question on answers.ros.org and then post the link here, and I'll get to it soon.

https://answers.ros.org/question/414956/fusing-imu-and-encoder-not-working-for-me/

foxydevelby, 2023-05-04:
I'm thinking If I set the publish_acceleration: to True how can I feed the topic /accel/filtered to navigation2 stack?