cra-ros-pkg/robot_localization #829: Docs and outdated link to answers.ros.org

christophfroehlich, 2023-08-25:
Hi! I'm just wondering, if there is a published version of your docs for ROS 2?
https://index.ros.org/p/robot_localization/ links only to http://wiki.ros.org/robot_localization linking to http://docs.ros.org/en/noetic/api/robot_localization/html/configuring_robot_localization.html with noetic only.

And just a little thing, in the repo description you guide people to answers.ros.org, which is not available any more.

ayrton04, 2023-08-25:
At the moment, no. There's no `doc` job [for the package ](https://github.com/ros/rosdistro/blob/48793b2f6f8e06ce5b661b9790ef0d925bd8d9d5/iron/distribution.yaml#L4869-L4880)on the build farm. I can add that stanza to [match ROS 1 distros](https://github.com/ros/rosdistro/blob/48793b2f6f8e06ce5b661b9790ef0d925bd8d9d5/noetic/distribution.yaml#L8708-L8711), but I am not sure if the ROS 2 build system uses the same documentation infrastructure that is used in ROS 1. Anyone know the answer to that? @SteveMacenski?

mwopfner, 2026-02-26:
Where to find the newest documentation. The newest I could find is 2.7.7 but the project is currently at 3.5.1
Unfortunately the _API Docs_ link at https://index.ros.org/p/robot_localization/ is broken and points to http://docs.ros.org/en/jazzy/p/robot_localization