cra-ros-pkg/robot_localization #839: UKF position data closely match slam absolute pose regard wheel odometry.

spectra543, 2023-10-28:
Hello, I have a specific question related to a robot localization package. I've been trying to get an answer on the Robot Stack Exchange for quite some time now, but I haven't received a response. I'm wondering if you could please take a moment to review my situation. 

I'm using the Unscented Kalman Filter (UKF) in the package to fuse the absolute pose from SLAM with the linear velocity from wheel odometry. However, I've noticed that the position output of the Kalman filter appears to closely match the SLAM absolute pose, especially in terms of the position data provided by the wheel odometry and the covariance matrix values.

 My question is whether this behavior is expected or if I might have made an error in my configuration. Additionally, I've observed that even when the SLAM data is cut off, the filter output continues to update based on data from the wheel odometry.

configuration file
`frequency: 30

sensor_timeout: 0.1

two_d_mode: true

transform_time_offset: 0.0

transform_timeout: 0.0

print_diagnostics: true

debug: false

debug_out_file: /path/to/debug/file.txt

map_frame: map              # Defaults to "map" if unspecified
odom_frame: odom            # Defaults to "odom" if unspecified
base_link_frame: base_link  # Defaults to "base_link" if unspecified
world_frame: odom           # Defaults to the value of odom_frame if unspecified

# Further input parameter examples
odom0: /odom_cov_ukf  # wheel odometry
odom0_config:  [false, false, false,
                false, false,false,
                true, true,  false,
                false, false, false,
                false, false, true]
odom0_differential: false
odom0_relative: true
odom0_queue_size: 3
#odom1_pose_rejection_threshold: 0.01
#odom1_twist_rejection_threshold: 0.2
odom0_nodelay: false

odom1: /cam_aru_odom
odom1_config: [false,false, false,
                false, false,false,
                true, true,  false,
                false, false, false,
                false, false, true]
odom1_differential: false
odom1_relative: false
odom1_queue_size: 3
#odom1_pose_rejection_threshold: 0.01
#odom1_twist_rejection_threshold: 0.2
odom1_nodelay: false

pose0: /vo_ukf_cov  #slam pose
pose0_config: [true,  true,  false,
               false, false, true,
               false, false, false,
               false, false, false,
               false, false, false]
pose0_differential: false
pose0_relative: fasle
pose0_queue_size: 1
#pose0_rejection_threshold: 2  # Note the difference in parameter name
pose0_nodelay: false

imu0: /imu_cov_data_ukf
imu0_config: [false, false, false,
             false,  false, false,
[... 96 lines cut ...]
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 6.08298869338e-05
      w: 1.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.00142620096449
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 6.862781704086771e-06]
......................................................................................................................
here's the picture to show you what's confuse me this picture show the position in x axis in straight run collected from my robot the red graph is from slam, green is wheel odometry, blue is from the robot_localzation. The Kalman filter seem to always follow the slam position output is this normal behavior? thank you so much for looking into my question. 
[image]

ayrton04, 2023-10-29:
Please ask questions on robotics.stackexchange.com

closed 2023-10-29 by ayrton04

spectra543, 2023-10-29:
Alright thanks already done [https://robotics.stackexchange.com/questions/105045/ukf-position-data-closely-match-slam-absolute-pose-regard-wheel-odometry](url)

spectra543, 2023-10-29:
> Please ask questions on robotics.stackexchange.com

Alright thanks already done [https://robotics.stackexchange.com/questions/105045/ukf-position-data-closely-match-slam-absolute-pose-regard-wheel-odometry](https://github.com/cra-ros-pkg/robot_localization/issues/url)

spectra543, 2023-10-29:
> Please ask questions on robotics.stackexchange.com

I already edit the code in stack exchange to be readable my apologies i'm quite  new to it

ayrton04, 2023-10-29:
Great, thanks. I'd ask for your patience, in that case.