cra-ros-pkg/robot_localization #859: Separating the msgs from the package

rafal-gorecki, 2024-01-08:
If it would not be a big problem for you, I would ask you to split the package into `robot-localization-msgs` and `robot-localization`.

Why?
When exchanging messages on multiple devices, it is necessary to download the entire package, which weighs a lot more than the services needed for reading and sending.

Regards

ayrton04, 2024-01-08:
This package is being deprecated, so I am reluctant to make that large a change to it.

You could always create your own package with a message structure that is identical to the messages in this package. The MD5 sums will be the same, so you'll be able to receive the message on the other end.

rafal-gorecki, 2024-01-18:
Thank you for your answer

closed 2024-01-18 by rafal-gorecki