cra-ros-pkg/robot_localization #871: State and Covariance blow up when a sensor update is delayed/missed.

lowellm91, 2024-03-20:
# ALL QUESTIONS SHOULD BE DIRECTED TO robotics.stackexchange.com

**Describe the bug**

When fusing multiple odometry sources using differential mode, when an output is missed the state transition matrix, its jacobian and the updated state blow up. Thresholding doesn't seem to filter this out.  

**To Reproduce**
Steps to reproduce the behavior:

I can't provide the bag file and odometry setup that causes this but basically an odometry source being missed causes system explosion. 

**Expected behavior**

I expected the extreme output to be filtered out via a threshold and the system to update using the existing velocity information. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntut 22.04
 - ROS Distribution: Humble
 - `robot_localization` Package Version:3.5.2

ayrton04, 2024-03-20:
This should be a question. As the template suggests, please direct questions to robotics.stackexchange.com.

closed 2024-03-20 by ayrton04

lowellm91, 2024-03-22:
Hi, I believe this is a bug, we think it's related to the system shifting between sim_time and real_time when the SLAM node isn't updated fast enough. The thing is even after setting all our nodes to use_sim_time, the problem still persists with the system blowing up. Do you have a suggestion in terms of tracking down if it's a bug?