cra-ros-pkg/robot_localization #872: EKF filter node not working correcty

jesusramondovale, 2024-03-20:
Hi! I'm using a two-fixed-wheeled differential robot with Nav2 on ROS2 Humble, I succesfully achieved to code a standard odom publisher which publishes on /odom and /tf topics like this: 

```
void OdomPublisherNode::timerCallback() 
{
  // Calculate current time and dt
  current_time = now();
  double dt = (current_time - last_time).seconds();

  // Calculate delta x, delta y and delta th
  double delta_x = (vx * cos(th) - vy * sin(th)) * dt;
  double delta_y = (vx * sin(th) + vy * cos(th)) * dt;
  double delta_th = vth * dt;
  x += delta_x;
  y += delta_y;
  th += delta_th;

  //Only for logging
  degrees = fmod((th * 180.0)/M_PI,360);
  RCLCPP_DEBUG(this->get_logger(), "Position on X: %.2f | Position on Y: %.2f | Orientation (TH) : %.2f º", x , y, degrees );
  
  // Create TransformStamped
  geometry_msgs::msg::TransformStamped odom_trans;
  odom_trans.header.stamp = current_time;
  odom_trans.header.frame_id = "odom";
  odom_trans.child_frame_id = "base_link";

  odom_trans.transform.translation.x = x;
  odom_trans.transform.translation.y = y;
  odom_trans.transform.translation.z = 0.0;

  // Create tf2::Quaternion
  tf2::Quaternion tf2_quat;
  tf2_quat.setRPY(0, 0, th);  // RPY: Roll, Pitch, Yaw

  // Create geometry_msgs::msg::Quaternion and fill
  geometry_msgs::msg::Quaternion odom_quat;
  odom_quat.x = tf2_quat.x();
  odom_quat.y = tf2_quat.y();
  odom_quat.z = tf2_quat.z();
  odom_quat.w = tf2_quat.w();
  odom_trans.transform.rotation = odom_quat;

  // Publish tf
  tf_broadcaster->sendTransform(odom_trans);

  // Create odom msg
  nav_msgs::msg::Odometry odom;
  odom.header.stamp = current_time;
  odom.header.frame_id = "odom";

  // Fill position and orientation and speed
  odom.pose.pose.position.x = x;
  odom.pose.pose.position.y = y;
  odom.pose.pose.position.z = 0.0;
  odom.pose.pose.orientation = odom_quat;  
  odom.child_frame_id = "base_link";
  odom.twist.twist.linear.x = vx;
  odom.twist.twist.linear.y = vy;
  odom.twist.twist.angular.z = vth;

  // Publish odometry on /odom
  odom_pub->publish(odom);

  tf2_msgs::msg::TFMessage tf_message;
  tf_message.transforms.push_back(odom_trans);

  last_time = current_time;
}
```
[... 98 lines cut ...]
initial_estimate_covariance: [1e-9,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    1e-9,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    1e-9,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    1e-9,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    1e-9,   0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    1e-9,   0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    1e-9,   0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    1e-9,   0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    1e-9,   0.0,     0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    1e-9,    0.0,     0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     1e-9,    0.0,     0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     1e-9,    0.0,    0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     1e-9,   0.0,    0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    1e-9,   0.0,
                                      0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    1e-9]

```

**Expected behavior**
I expected that the EKF filter node smoothed something on its /tf and /odom publications but it is not doing it.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS Distribution: Humble
 - `robot_localization` Package Version: 3.5.2

ayrton04, 2024-03-20:
Thanks for the question. However, as mentioned in the issue template, all questions should be directed to `robotics.stackexchange.com`. Thanks.

closed 2024-03-20 by ayrton04

jesusramondovale, 2024-03-20:
Okey thank you and sorry