cra-ros-pkg/robot_localization #873: When floating point values are used as a time-measurement, delta is incorrectly scaled for state-propagation.

lowellm91, 2024-03-22:
# ALL QUESTIONS SHOULD BE DIRECTED TO robotics.stackexchange.com

## When asking a question on robotics.stackexchange.com, please include your full EKF config and one sample message from each sensor input**

# THIS PACKAGE WILL SOON BE DEPRECATED. UNLESS SEVERE, BUG REPORTS ARE UNLIKELY TO BE ADDRESSED

**Describe the bug**
1. When our SLAM systems fail to update fast enough the EKF filter must propagate the state itself, we have found out that in this case the delta value is not correctly scaled when it's a floating point number.  

**To Reproduce**
Steps to reproduce the behavior:
1. Cannot provide bag files or setup as they are confidential. 

**Expected behavior**
A clear and concise description of what you expected to happen.

1. Under normal state propagation, the time step value is scaled to seconds. We have found that when the delta value is a floating point number that scaling is not executed correctly.  Look at the transfer function screen shots starting at entry (1,7) and going along the off diagonals for both pictures

**Screenshots**

Correct propagation:
[image]

Faulty propagation:

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS Distribution: Humble
 - `robot_localization` Package Version:3.5.2

**Additional context**
Add any other context about the problem here.

lowellm91, 2024-03-22:
I believe the error is some place in the utility function which converts nanoseconds to seconds but I could be wrong.

ayrton04, 2024-03-22:
If you believe there is a bug, you need to provide a reproducible example. You can appreciate that posting a vague issue with no concrete steps to reproduce is not very helpful.

In any case, I don't have the cycles to guess at your issue. If you can find a way to reproduce it outside your proprietary system, I can try to find time to take a look.

ayrton04, 2024-03-22:
By the way, are you only fusing one sensor data source into the filter? And how long a time delta are we talking? I fully expect the state estimate covariance to explode when you don't measure a given pose variable or its velocity for a period.

lowellm91, 2024-03-22:
Less than .5 seconds, the screenshots show where the failure is. The issue is the node is incorrectly converting nanoseconds to seconds resulting in the nanosecond value being used as the time step. I think it's related to the nanosecond value being a float.

lowellm91, 2024-03-23:
Alright I tracked down the issue, I thought it was in relation to floating point stuff because outputs were correct some of the time but this was only because prediction was off. The issue comes down to:

rclcpp::Time cur_time = this->now();

Which doesn't get the bag files time and instead gets the system time, I tried launching it with use_sim_time but it doesn't seem to make a difference, suggestions?

lowellm91, 2024-03-23:
Problem still exists but since it's caused not by what I thought was the issue, will close issue. Do you have any idea why it's not using sim time though?

closed 2024-03-23 by lowellm91

ayrton04, 2024-03-25:
According to all the ROS 2 documentation, if `use_sim_time` is true, the `Node` class should automatically switch to ROS time. The source code for `rclcpp` appears to confirm that:

https://github.com/ros2/rclcpp/blob/b50f9ab41840dfec121b7d12118da6e6593d21ae/rclcpp/src/rclcpp/time_source.cpp#L265-L267

Are you sure the parameter is being set correctly? You could always clone the package, build it in your workspace, and add a check here to see if that parameter is set:

https://github.com/cra-ros-pkg/robot_localization/blob/1cb953379378d13b3e0935ed73d0c7407bd80543/src/ros_filter.cpp#L115

```
if (this->has_param("use_sim_time"))
{
  RCLCPP_INFO(this->get_logger(), "Sim time parameter exists.");
  // Add some checks for the actual _value_ of use_sim_time here
}
else
{
  RCLCPP_ERROR(this->get_logger(), "Sim time parameter does not exist");
}
```

lowellm91, 2024-03-26:
By adding a hard limit to the delta in source code we got expected behavior, the issue is 100% system time being used over bag time when sensor timeout happens. I verified use_sim_time was set for every node, verified subscribers etc. I don't understand why use_sim_time isn't working but it's not.