cra-ros-pkg/robot_localization #874: New iron release

Timple, 2024-03-25:
Last iron release seems 10 months ago. But still some [nice stuff](https://github.com/cra-ros-pkg/robot_localization/compare/3.6.0...ros2)  including the `SetUTMZone.srv` have been added and forward-ported.

Would one be so kind to do a bloom release for iron?

ayrton04, 2024-04-16:
Your link is comparing `ros2` (`rolling`) to tag 3.6.0 (also used in the `rolling` [distro](https://github.com/ros/rosdistro/blob/8858a76001dc0228c91658939f947fe708c10bf0/rolling/distribution.yaml#L5530)). Are you wanting a new `rolling` release or `iron`? The `iron` release is [3.5.0](https://github.com/ros/rosdistro/blob/8858a76001dc0228c91658939f947fe708c10bf0/iron/distribution.yaml#L5573), which is closer to `humble` than `rolling`.

I'm happy to move the `iron` release to be the same as `rolling`, though I'd probably do a minor bump so that `iron` has its own version and isn't piggy-backing off of `rolling` releases.

Timple, 2024-04-16:
I was hoping on a new iron release. I would indeed make it the same as rolling. As it works just fine.

ayrton04, 2024-04-16:
OK, I went on a "bloom spree" and hit every active distro with changes.

closed 2024-04-16 by ayrton04

Timple, 2024-04-16:
Awesome 😎