cra-ros-pkg/robot_localization #877: Issues when fusing odometry for Ackermann steered robot

Rak-r, 2024-04-03:
Platform: ROS2 Humble,

I have opened a question on robotics stack exchange. It would be great if somebody could provide any guidance to deal with the problem. 

Reference link: https://robotics.stackexchange.com/questions/110374/issues-while-setting-up-ekf-for-ackermann-steered-robot 
@ayrton04

ayrton04, 2024-04-03:
Please don't cross post. I will look at your issue when I have time.

closed 2024-04-03 by ayrton04

Rak-r, 2024-04-08:
Could you provide any insight about the issue mentioned in the above link. I have been reading about similar setups and trying  but getting a bit confused. @ayrton04

ayrton04, 2024-04-08:
> I will look at your issue when I have time.

I'm afraid this is still true. I will get to it when I can. Please don't keep asking.