cra-ros-pkg/robot_localization #883: navsat_transform_node cannot work

sunshineharry, 2024-05-03:
Hi~

Thank you very much!

I am using the navsat_transform_node. However, it do not work as expected.

The loggers are the same and shown in screen are as follows:

```
[INFO] [1714719602.692670968] [navsat_transform_node]: Datum UTM coordinate is (50S, 450644.47, 4123950.69)
[navsat_transform_node-1] [INFO] [1714719602.742924109] [navsat_transform_node]: Datum (latitude, longitude, altitude) is (49.96, 126.31, 51.82)
```

**However, the topics of `/gps/filtered`, `odometry/filtered`, `odometry/gps` have not been published ! It can be shown in `ros2 topic list`, but it has no data !**

Here is something about my system

- System: Ubuntu2204
- ROS Version: ROS Humble

Here is the configuration of the navsat_transform_node:

- launch file:

```python
def generate_launch_description():
    gps_config = LaunchConfiguration('gps_config')

    return LaunchDescription([
        DeclareLaunchArgument(
            default_value=[FindPackageShare('rtk_imu_fusion'), '/config/gps_config.yaml'],
            description='gps_config',
        ),

        Node(
            package='robot_localization',
            executable='navsat_transform_node',
            name='navsat_transform_node',
            parameters=[gps_config], 
            remappings=[
                    ('/gps/fix', 'sensors/gps/fix'),
                    ('/imu',     '/sensors/imu'),
                ],
        )
    ])
```

- configuration file
```
navsat_transform_node:
  ros__parameters:
    frequency: 10.0
    magnetic_declination_radians: 0.13282
    yaw_offset: 0.0
    zero_altitude: false
    publish_filtered_gps: true
    broadcast_utm_transform: true
    transform_timeout: 0.0
```

Some ROS topics are shown as follows:
- `sensors/gps/fix`
```
header:
  stamp:
    sec: 1714719620
    nanosec: 91937117
  frame_id: world
status:
  status: 2
  service: 1
latitude: 49.96362052716667
longitude: 117.30501391933333
altitude: 31.831900000000005
position_covariance:
- 0.000144
- 0.0
- 0.0
- 0.0
- 0.000144
- 0.0
- 0.0
- 0.0
- 0.002304
position_covariance_type: 1
```

- `/sensors/imu`
```
header:
  stamp:
    sec: 1714719639
    nanosec: 620371315
  frame_id: imu
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 1.0
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.01
angular_velocity:
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.00439453125
  y: 0.0029296875
  z: 0.9931640625
linear_acceleration_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
```

Thank you very much !!!

ayrton04, 2024-05-03:
Thanks for the question. However, as the issue template states, all questions should be directed to robotics.stackexchange.com.

closed 2024-05-03 by ayrton04

peakyquest, 2025-04-02:
I got the same issue. Have you solve the issue.

peakyquest, 2025-12-17:
Yes I have solved the issue, I wasnt using the correct values for navsat transformation.