cra-ros-pkg/robot_localization #887: QoS incompatibility between the navsat_transform node and the ekf_filter node

GPrathap, 2024-06-09:
**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS Distribution: Humble 
 
To reproduce this behaviour 

1.  `ros2 topic info -v /odometry/global`  where `/odometry/global` is the global odometry estimation topic  from `ekf_filter`

     
            Publisher count: 1
            Node name: ekf_filter_node_map
            Node namespace: /
            Topic type: nav_msgs/msg/Odometry
            Endpoint type: PUBLISHER
            GID: 01.10.90.de.c2.02.9b.31.57.de.c5.4d.00.00.28.03.00.00.00.00.00.00.00.00
            QoS profile:
              Reliability: RELIABLE
              History (Depth): KEEP_LAST (10)
              Durability: VOLATILE
              Lifespan: Infinite
              Deadline: Infinite
              Liveliness: AUTOMATIC
              Liveliness lease duration: Infinite

     Subscription count: 2

        Node name: navsat_transform_base
        Node namespace: /
        Topic type: nav_msgs/msg/Odometry
        Endpoint type: SUBSCRIPTION
        GID: 01.10.42.3d.a3.e1.64.b1.da.72.99.8a.00.00.22.04.00.00.00.00.00.00.00.00
        QoS profile:
          Reliability: BEST_EFFORT
          History (Depth): KEEP_LAST (1)
          Durability: VOLATILE
          Lifespan: Infinite
          Deadline: Infinite
          Liveliness: AUTOMATIC
          Liveliness lease duration: Infinite

I feel that within ` ekf_filter`  node for the odometry topic QoS type should be BEST_EFFORT, VOLATILE.

GPrathap, 2024-06-10:
@ayrton04 If we configure this way will it solve the problem? https://docs.ros2.org/galactic/api/rclcpp/classrclcpp_1_1QosOverridingOptions.html

GPrathap, 2024-06-10:
Seems to be when I set this it working fine :)

        ekf_filter_node_map:
          ros__parameters:
            qos_overrides:
              /odometry/global:
                publisher:  # publisher without provided id
                  reliability: best_effort
                  depth: 1

ayrton04, 2024-06-24:
I don't use this node in ROS 2, so I don't feel strongly. Is it actually causing any performance issues?

GPrathap, 2024-06-24:
Yep, over DDS, especially with Cyclone DDS. But, after ordering it's fine

ayrton04, 2024-06-24:
And why change the EKF settings and not the QoS profile in navsat_transform_node?

ayrton04, 2024-06-24:
Okie doke. I won't be making any changes, but feel free to submit a PR. Thanks.