cra-ros-pkg/robot_localization #890: High CPU Usage with navsat_transform Node in ROS 2 Jazzy

porizou, 2024-06-22:
I am experiencing an issue where the navsat_transform node in ROS 2 Jazzy consistently causes the CPU usage to exceed 100%. This occurs during normal operation and significantly impacts system performance. Here are the details of my setup:

System Information:

- OS: Ubuntu 24.04LTS
- ROS 2 : Jazzy
- navsat_transform package version: 3.8.0

parameter of navsat_transform_node

```yaml
navsat_transform_node:
  ros__parameters:
    frequency: 10.0
    delay: 3.0
    yaw_offset: 1.57
    magnetic_declination_radians: 0.0
    zero_altitude: true
    broadcast_utm_transform: false
    publish_filtered_gps: false
    use_odometry_yaw: false
    wait_for_datum: false
```

[image]

ayrton04, 2024-06-24:
Thanks for the report. Is this with no messages publishing from any other nodes, i.e., you are just running this in isolation?

porizou, 2024-07-01:
Thank you for your response. Here are some additional details regarding my setup:

- I am running the navsat_transform_node in combination with the ukf_node.
- The GPS NavSatFix topic is being obtained from Gazebo Sim through the gz_bridge.
- When I run only the navsat_transform_node using the command ros2 run robot_localization navsat_transform_node, the CPU usage does not reach 100%.

This is rqt_graph.
CPU usage of navsat_transform is at 100%, but the node itself is functioning normally.

[image]

ayrton04, 2024-08-26:
If you could provide a full config and bag file (i.e., everything I'd need to replicate the issue), that would be helpful, thank you.

porizou, 2024-09-18:
Hi,
The default value of transform_timeout was 0.0, so setting a value other than 0.0 solved the problem.

```param.yaml
navsat_transform_node:
  ros__parameters:
    frequency: 5.0
    delay: 3.0
    yaw_offset: 0.0
    magnetic_declination_radians: 0.0
    zero_altitude: true
    broadcast_utm_transform: false
    publish_filtered_gps: false
    use_odometry_yaw: false
    wait_for_datum: false
    transform_timeout: 0.5
```

maxhochlenertarnoldnextg, 2024-11-21:
Having the same issue