cra-ros-pkg/robot_localization #891: Incorrect UTM to Odom local coordinates (x,y) and Orientation Conversion with Magnetic Declination and Yaw Offset in São Paulo, Brazil (Real Robot)

marcusvinicius178, 2024-07-03:
I am working on an autonomous vehicle project in  **São Paulo, Brazil**. As part of the project, I am using the **navsat_transform_node** from the ROS robot_localization package for integrating GPS data. I have calculated the magnetic declination for my location using the NCEI site:[NCEI Magnetic Calculators]( https://www.ngdc.noaa.gov/geomag/calculators/magcalc.shtml) The declination value is below:

[image]

Our IMU yaw rate is obtained from our vehicle's CAN yaw rate data, and we integrate that value. We don't have a direct IMU source.

**Problem:**

Despite correctly setting the magnetic declination and trying various values for the yaw_offset, **my robot's orientation on Mapviz is consistently incorrect**. The robot's **arrow points to the back instead of the front**, resulting in incorrect path planning. Specifically, the robot is oriented between 90 - 180 degrees opposite to its actual heading. Navsat is also **making errors converting to lat,long to UTM** for South Hemisphere:  

`[navsat_transform_node-6]   what():  Easting 3.12708e+06km not in UTM range for S hemisphere [0km, 1000km]
`
**Configuration Attempts:**
I've tried different values as you can check on my configuration file [ https://pastebin.com/sT3LZV0g](dual_ekf_navsat_params.yaml)

Magnetic Declination: -0.38201051083 radians (as calculated)

yaw_offset values tried: pi, -pi, 2pi, -2pi, etc.

None of these attempts have corrected the orientation issue. Here is a snippet of my navsat_transform_node configuration:

```
    navsat_transform:
      ros__parameters:
        use_sim_time: false
        frequency: 30.0
        delay: 3.0
        magnetic_declination_radians: -0.38201051083
        yaw_offset: 3.14159265359  # Example value, tried several others
        zero_altitude: true
        broadcast_utm_transform: true
        publish_filtered_gps: true
        use_odometry_yaw: false
        wait_for_datum: true 
        datum: [-23.66075725, -46.5925, 0.0]
```

**Errors and Warnings:**

Additionally, I am encountering the following error on the terminal, which suggests a possible issue with the UTM conversion for the southern hemisphere:

```
    [mapviz-15] Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1719317715,336261 according to authority Authority undetectable
    [mapviz-15] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
    [mapviz-15]          at line 292 in ./src/buffer_core.cpp
    [navsat_transform_node-6] terminate called after throwing an instance of 'GeographicLib::GeographicErr'
    [navsat_transform_node-6]   what():  Easting 3.12708e+06km not in UTM range for S hemisphere [0km, 1000km]
    [ERROR] [navsat_transform_node-6]: process has died [pid 17979, exit code -6, cmd '/opt/ros/iron/lib/robot_localization/navsat_transform_node --ros-args -r __node:=navsat_transform --params-file /home/rota_2024/nav2_gps_ws/install/nav2_bringup/share/nav2_bringup/params/real_bot_dual_ekf_navsat_params.yaml --params-file /tmp/launch_params_n1pfb_qu -r imu/data:=imu/data -r gps/fix:=gps/fix -r gps/filtered:=gps/filtered -r odometry/gps:=odometry/gps -r odometry/filtered:=odometry/global'].
```

**Questions:**

 1. **Magnetic Declination Usage:** Is there anything I might be missing or
    incorrectly setting regarding the magnetic_declination_radians?
    Should this value be used differently? 
 2. **Yaw Offset Adjustment:** How
    can I correctly determine the yaw_offset to align my robot's heading
    in the correct direction? Any specific steps or calculations to
    ensure the correct value? 
 3. **UTM Range Error:** What could be causing the
    GeographicLib::GeographicErr related to UTM range, and how can it be
    resolved, especially for locations in the southern hemisphere?

**Additional Information:**

I am using ROS 2 Iron.
I will attach a video and pictures demonstrating the issue for better visualization.

[Mapviz Orientation Issue Video](https://www.youtube.com/watch?v=ZGIMscIPAWk)

Wrong Orientation Picture:

[image]

Desired Orientation (Robot Forward Front)

[image]
Insights are welcome! 

I have checked other users with similar issue: https://robotics.stackexchange.com/questions/112051/localization-orientation-issue-while-running-robot-localization-with-navsat-tran

I have also opened this issue on robotics stack exchange, but unfortunately didn't have an assistance there: https://robotics.stackexchange.com/

  [1]: https://www.ngdc.noaa.gov/geomag/calculators/magcalc.shtml
  [2]: https://i.sstatic.net/jy8yE4vF.png
  [3]: https://pastebin.com/sT3LZV0g
  [4]: https://youtu.be/ZGIMscIPAWk
  [5]: https://i.sstatic.net/r7PxHykZ.png
  [6]: https://i.sstatic.net/KZh1jLGy.png

ayrton04, 2024-07-03:
Thanks for the detailed question. However, all questions should go on robotics.stackexchange.com. I see you've opened the question there, and to prevent cross-posting, I will close this.

Please note that I have very little free time to dedicate to this package these days, so I'll answer the question as soon as I have time to catch up.

closed 2024-07-03 by ayrton04

marcusvinicius178, 2024-07-03:
Thanks @ayrton04 sorry to publish question here. I just wished to caught your attention, since I have read numerous good answers and solutions you have provided to the community!You might be the unique expert that can help me, I am not a localization expert. Take you time to answer me! Thanks very much and sorry for eventual disturbance.