cra-ros-pkg/robot_localization #902: EKF filter causes jumpy odom (only encoder based). Without any warning or error.

XantaKross, 2024-09-10:
# ALL QUESTIONS SHOULD BE DIRECTED TO robotics.stackexchange.com

## When asking a question on robotics.stackexchange.com, please include your full EKF config and one sample message from each sensor input**

# THIS PACKAGE WILL SOON BE DEPRECATED. UNLESS SEVERE, BUG REPORTS ARE UNLIKELY TO BE ADDRESSED

**Describe the bug**
Please help me. I've been trying to fix this error for a long time without any luck.
I'm using this stack for my differential drive robot. Similar to the one built here: Articulated Robotics (https://www.youtube.com/playlist?list=PLunhqkrRNRhYAffV8JDiFOatQXuU-NnxT)
1. diffDrive-Arduino: https://github.com/joshnewans/diffdrive_arduino/tree/humble
2. ros2 mapper
3. robot_localization

This stack works very well in simulation (gazebo) to give robust reading of odom. However, when I use this in the real robot I see a bizzare jump when I map out a room and come back to my initial /map tf at (0, 0) from where I started.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the basic robot with online_async mapper.
2. Run the EKF filter I've used.
3. Try moving the robot a bit far away using teleop then come back to the /map tf.

**Expected behavior**
The robot_localization should be stable and mustn't cause any jumps. Especially as the stack works robustly with nav2 if I don't use
robot_localization or launch ekf filter. 

**Screenshots**
I have added a video to better understand my error as screenshots won't be enough.
https://youtu.be/_7vcquzLZYU

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS
 - ROS Distribution: Humble
 - `robot_localization` Package Version: Latest.

**Additional context**

I've observed the following things:
1. the odom0 by itself without anything else is enough to cause this error in ekf filter. Irrespective of whether there is any imu0 or pose0.
2. the jump doesn't occur if I don't use ekf filter. I can run it fine with just publishing the odom from the /diff_cont/odom topic which is published using encoder data from arduino. (REFERENCE: https://www.youtube.com/watch?v=4QKsDf1c4hc)
3. I actually observed way bigger jumps a while ago. But I was able to stop that big jump by altering the tfs being published. So my guess now would be something like that is happening here again? I don't really know.
4. The error seems to be timing or space based. As if I run for some time only then does it show these bugs, not immediately.

ayrton04, 2024-09-10:
Thanks for the question. However, as per the issue template that you pasted in the question itself, all questions should be directed to robotics.stackexchange.com. If you've already asked there, please be patient, as I get very little time to do support for this package. Thanks.

ayrton04, 2024-09-10:
Also, when you ask on Stack Exchange, please include your full EKF config and a sample message from each sensor input.

closed 2024-09-10 by ayrton04