cra-ros-pkg/robot_localization #903: ros2 param dump does not work

reinzor, 2024-09-17:
**Describe the bug**

`ros2 param dump` is not dumping parameters for the robot localization nodes.

**To Reproduce**
Steps to reproduce the behavior:
1. `ros2 run robot_localization ekf_node`
2. `ros2 param dump /ekf_filter_node` (using https://github.com/ros2/ros2cli/pull/933)

```
/ekf_filter_node:
  ros__parameters: {}
```

**Expected behavior**

Should dump the parameters of the node.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: 24.04
 - ROS Distribution: Jazzy
 - `robot_localization` Package Version:  3.8.0

**Additional context**

`ros2 param list /ekf_filter_node` works:

```
  base_link_frame
  base_link_frame_output
  control_timeout
  debug
  diagnostic_updater.period
  diagnostic_updater.use_fqn
  disabled_at_startup
  dynamic_process_noise_covariance
  frequency
  gravitational_acceleration
  history_length
  imu0
  initial_estimate_covariance
  initial_state
  map_frame
  odom0
  odom_frame
  permit_corrected_publication
  pose0
  predict_to_current_time
  print_diagnostics
  process_noise_covariance
  publish_acceleration
  publish_tf
  qos_overrides./odometry/filtered.publisher.depth
  qos_overrides./odometry/filtered.publisher.history
  qos_overrides./odometry/filtered.publisher.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  reset_on_time_jump
  sensor_timeout
  smooth_lagged_data
  start_type_description_service
  tf_prefix
  transform_time_offset
  transform_timeout
  twist0
  two_d_mode
  use_control
  use_sim_time
  world_frame
```

Single value getting works:

`ros2 param get /ekf_filter_node base_link_frame`:

```
String value is: base_link
```

Related to https://github.com/ros2/ros2cli/issues/930

SteveMacenski, 2024-09-17:
Dumping params don't work when there are no defaults set for a given parameter  (i.e. set as `PARAMETER_TYPE_DOUBLE` rather than given a default of `1.0` or a real value). I think this is just an instantiation of that and isn't a bug in R_L but a bug in the CLI tool itself that it cannot handle unset but declared variables. I've run into this kind of thing before in Nav2.

reinzor, 2024-09-18:
Thanks for the clarification @SteveMacenski . Though I am not sure whether the problem is in the CLI. Since getting individual parameters works but getting the full list doesn't (that is was `ros2 param dump` does). So might even be in the client library layer.

SteveMacenski, 2024-09-19:
Individual parameters would work, if you're not querying parameters that have no default value set. I'm guessing that the EKF node has at least one such parameter so it fails and then returns nothing. I've seen that happen before with dynamic parameter updates

reinzor, 2024-09-20:
`ros2 param get` for all individual parameters:

```
echo "base_link_frame: $(ros2 param get /ekf_filter_node base_link_frame)"
echo "base_link_frame_output: $(ros2 param get /ekf_filter_node base_link_frame_output)"
echo "control_timeout: $(ros2 param get /ekf_filter_node control_timeout)"
echo "debug: $(ros2 param get /ekf_filter_node debug)"
echo "diagnostic_updater.period: $(ros2 param get /ekf_filter_node diagnostic_updater.period)"
echo "diagnostic_updater.use_fqn: $(ros2 param get /ekf_filter_node diagnostic_updater.use_fqn)"
echo "disabled_at_startup: $(ros2 param get /ekf_filter_node disabled_at_startup)"
echo "dynamic_process_noise_covariance: $(ros2 param get /ekf_filter_node dynamic_process_noise_covariance)"
echo "frequency: $(ros2 param get /ekf_filter_node frequency)"
echo "gravitational_acceleration: $(ros2 param get /ekf_filter_node gravitational_acceleration)"
echo "history_length: $(ros2 param get /ekf_filter_node history_length)"
echo "imu0: $(ros2 param get /ekf_filter_node imu0)"
echo "initial_estimate_covariance: $(ros2 param get /ekf_filter_node initial_estimate_covariance)"
echo "initial_state: $(ros2 param get /ekf_filter_node initial_state)"
echo "map_frame: $(ros2 param get /ekf_filter_node map_frame)"
echo "odom0: $(ros2 param get /ekf_filter_node odom0)"
echo "odom_frame: $(ros2 param get /ekf_filter_node odom_frame)"
echo "permit_corrected_publication: $(ros2 param get /ekf_filter_node permit_corrected_publication)"
echo "pose0: $(ros2 param get /ekf_filter_node pose0)"
echo "predict_to_current_time: $(ros2 param get /ekf_filter_node predict_to_current_time)"
echo "print_diagnostics: $(ros2 param get /ekf_filter_node print_diagnostics)"
echo "process_noise_covariance: $(ros2 param get /ekf_filter_node process_noise_covariance)"
echo "publish_acceleration: $(ros2 param get /ekf_filter_node publish_acceleration)"
echo "publish_tf: $(ros2 param get /ekf_filter_node publish_tf)"
echo "qos_overrides./odometry/filtered.publisher.depth: $(ros2 param get /ekf_filter_node qos_overrides./odometry/filtered.publisher.depth)"
echo "qos_overrides./odometry/filtered.publisher.history: $(ros2 param get /ekf_filter_node qos_overrides./odometry/filtered.publisher.history)"
echo "qos_overrides./odometry/filtered.publisher.reliability: $(ros2 param get /ekf_filter_node qos_overrides./odometry/filtered.publisher.reliability)"
echo "qos_overrides./parameter_events.publisher.depth: $(ros2 param get /ekf_filter_node qos_overrides./parameter_events.publisher.depth)"
echo "qos_overrides./parameter_events.publisher.durability: $(ros2 param get /ekf_filter_node qos_overrides./parameter_events.publisher.durability)"
echo "qos_overrides./parameter_events.publisher.history: $(ros2 param get /ekf_filter_node qos_overrides./parameter_events.publisher.history)"
echo "qos_overrides./parameter_events.publisher.reliability: $(ros2 param get /ekf_filter_node qos_overrides./parameter_events.publisher.reliability)"
echo "qos_overrides./tf.publisher.depth: $(ros2 param get /ekf_filter_node qos_overrides./tf.publisher.depth)"
echo "qos_overrides./tf.publisher.durability: $(ros2 param get /ekf_filter_node qos_overrides./tf.publisher.durability)"
echo "qos_overrides./tf.publisher.history: $(ros2 param get /ekf_filter_node qos_overrides./tf.publisher.history)"
echo "qos_overrides./tf.publisher.reliability: $(ros2 param get /ekf_filter_node qos_overrides./tf.publisher.reliability)"
echo "reset_on_time_jump: $(ros2 param get /ekf_filter_node reset_on_time_jump)"
echo "sensor_timeout: $(ros2 param get /ekf_filter_node sensor_timeout)"
echo "smooth_lagged_data: $(ros2 param get /ekf_filter_node smooth_lagged_data)"
echo "start_type_description_service: $(ros2 param get /ekf_filter_node start_type_description_service)"
echo "tf_prefix: $(ros2 param get /ekf_filter_node tf_prefix)"
echo "transform_time_offset: $(ros2 param get /ekf_filter_node transform_time_offset)"
echo "transform_timeout: $(ros2 param get /ekf_filter_node transform_timeout)"
echo "twist0: $(ros2 param get /ekf_filter_node twist0)"
echo "two_d_mode: $(ros2 param get /ekf_filter_node two_d_mode)"
echo "use_control: $(ros2 param get /ekf_filter_node use_control)"
echo "use_sim_time: $(ros2 param get /ekf_filter_node use_sim_time)"
echo "world_frame: $(ros2 param get /ekf_filter_node world_frame)"

base_link_frame: String value is: base_link
base_link_frame_output: String value is: base_link
control_timeout: Double value is: 0.0
debug: Boolean value is: False
diagnostic_updater.period: Double value is: 1.0
diagnostic_updater.use_fqn: Boolean value is: False
disabled_at_startup: Boolean value is: False
dynamic_process_noise_covariance: Boolean value is: False
frequency: Double value is: 30.0
gravitational_acceleration: Double value is: 9.80665
history_length: Double value is: 0.0
map_frame: String value is: map
odom_frame: String value is: odom
permit_corrected_publication: Boolean value is: False
predict_to_current_time: Boolean value is: False
print_diagnostics: Boolean value is: False
publish_acceleration: Boolean value is: False
publish_tf: Boolean value is: True
qos_overrides./odometry/filtered.publisher.depth: Integer value is: 10
[... 16 lines cut ...]
two_d_mode: Boolean value is: False
use_control: Boolean value is: False
use_sim_time: Boolean value is: False
world_frame: String value is: odom
imu0: Parameter not set
initial_estimate_covariance: Parameter not set
initial_state: Parameter not set
odom0: Parameter not set
[... cut ...] escription_service", "transform_time_offset", "transform_timeout", "two_d_mode", "use_control", "use_sim_time", "world_frame"]'
```

This will return a proper result.

However, when I add `odom0` to the list (parameter that was not set) it will give me an empty response. 

Not sure whether this is desired behavior. In any case, `ros2 param dump` should still work in my opinion if parameters exist that have no value set. There is also no way of retrieving this information using another interface.

ayrton04, 2024-09-20:
Would this not suggest an issue with `ros2 param dump`, rather than this package?

reinzor, 2024-09-20:
> Would this not suggest an issue with ros2 param dump, rather than this package?

Yes, I will close this issue and re-open one in https://github.com/ros2/ros2cli

Thanks all.

closed 2024-09-20 by reinzor

SteveMacenski, 2024-09-20:
I agree, I’d love to see this fixed. In any situation we have an array of parameters (i.e. don’t know its size, so we declare and check if set for the next) that node will have this kind of issue.