cra-ros-pkg/robot_localization #908: [Notice] Nav2 Migrated to TwistStamped to replace Twist for cmd_vel Topics

SteveMacenski, 2024-12-03:
Hi,

As a key community project using Nav2, I wanted to drop you a note that I am in the process of changing the default cmd_vel topic type from Twist to TwistStamped in efforts of enabling a broader set of applications and safety constraints. This has been an on-going discussion over the last 18 months and now has the technology stack in GZ, ROS Control, and Nav2 to enable it.

To migrate, your robot and simulation should soon accept a TwistStamped message rather than a Twist message. We have a couple of provided utilities that could make this easier for you, by switching to use a Twist Publisher and Twist Subscriber [1-2]. This is what is now used in all Nav2 nodes to either publish or subscribe to a Twist or TwistStamped message.

If you would like to continue to use a Twist, you can set enable_stamped_cmd_vel to false in all nodes in your configuration that use Twist Publisher/Subscriber objects, but this will be shortly switched to being default true.

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_publisher.hpp
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_subscriber.hpp

Nav2 migration:
- https://github.com/ros-navigation/navigation2/pull/4779
- https://github.com/ros-navigation/docs.nav2.org/pull/614
- https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/pull/16

SteveMacenski, 2024-12-03:
I think it would just be flipping the following line: https://github.com/cra-ros-pkg/robot_localization/blob/ef0a27352962c56a970f2bbeb8687313b9e54a9a/src/ros_filter.cpp#L975 by default and in the configuration files.

@ayrton04 I can do this -- are you OK with a change of this nature for `rolling` given no-new-features?

ayrton04, 2024-12-04:
Sure!

closed 2025-08-28 by ayrton04