cra-ros-pkg/robot_localization #921: New Jazzy release request

josephduchesne, 2025-03-12:
Can we get a new Jazzy release?

The parameter `use_control` is essentially useless in Jazzy due to the `stamped_control` parameter not working in the latest release (and Jazzy moving towards fully stamped messages most places). This was fixed in a PR merged 6 months ago, and the Jazzy documentation details this parameter, but it's still not out there.

Fix pr: https://github.com/cra-ros-pkg/robot_localization/pull/900

SteveMacenski, 2025-03-13:
@ayrton04 the last couple of times I tried to do releases I couldn't since I can't bypass code-review for my own PRs to open one and merge it with the cherry picked commits and updates to the version. 

I'm happy to run bloom as part of my regular scheduled software releases for Nav2, et al. but it might be easier for you to do this since that control issue blocks me from getting it all out in a sitting

ayrton04, 2025-03-20:
Yes, sorry, I will get on this today or tomorrow. Sorry for the delay.