cra-ros-pkg/robot_localization #924: GPS and IMU Fusion

peakyquest, 2025-04-02:
I am trying to get gps/odometry but the navsat_transform is not publishing anything. Below is my launch file. I am not sure I am using the navsat node properly. utm transform is not being published by the node. 

```
import launch
import launch_ros.actions
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument(
            'use_sim_time', default_value='true', description='Use simulation clock'
        ),

        launch_ros.actions.Node(
            package='robot_localization',
            executable='navsat_transform_node',
            name='navsat_transform_node',
            output='screen',
            parameters=[{
                'use_sim_time': LaunchConfiguration('use_sim_time'),
                'magnetic_declination_radians': 0.0,  
                'yaw_offset': 0.0,
                'zero_altitude': True,
                'broadcast_cartesian_transform': True,  # Updated parameter
                'publish_filtered_gps': False,
                'wait_for_datum': False,
                'use_odometry_yaw': False,  # Assuming IMU lacks orientation
            }],
            remappings=[
                ('/imu', '/wamv/sensors/imu/imu/data'),
                ('/gps/fix', '/wamv/sensors/gps/gps/fix'),
                ('/odometry/gps', '/odometry/gps')
            ]
        ),
    ])
``` 
This is my current tf tree
[image]

6itf1ip, 2025-04-03:
I am working on similar project have you solved the issue.

peakyquest, 2025-04-03:
I have also posed this in [robotstacks](https://robotics.stackexchange.com/questions/115243/how-to-use-robot-localization-using-gps-and-imu-only)