cra-ros-pkg/robot_localization #937: Odometry covariance off-diagonal elements in z row and column cause issues with EKF in two_d_mode

herolada, 2025-08-14:
**Describe the bug**
When using the EKF in `two_d_mode: true`, odometry messages (specifically `odometry/gps` from `navsat_transform_node`) which contain off-diagonal elements in positions related to the z coordinate (i.e. `cov[0,2]`; `cov[1,2]`; `cov[2,0]`; `cov[2,1]`) cause the localization to behave unpredictably eventually leading to complete divergence.

**To Reproduce**
Steps to reproduce the behavior:
1. Run a `navsat_transform` node with a GPS sensor which provides non-zero elements in covariance z-positions (in our case Septentrio mosaic-H).
2. Run a `ekf_localization` node with an `odometry/gps` input with `two_d_mode: true`. E.g.:

`odom0: odometry/gps 
    odom0_config: [true, true, false, 
                  false, false, false, 
                  false, false, false, 
                  false, false, false, 
                  false, false, false] 
    odom0_differential: false 
    odom0_queue_size: 10`

**Expected behavior**
With the two_d_mode enabled the odometry covariance elements in z-positions should not have any effect on the localization.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24
 - ROS Distribution: jazzy
 - `robot_localization` Package Version: 3.8.2

ayrton04, 2025-08-14:
This should probably be a question on robotics.stackexchange.com, but at the very least, you'll need to provide your full configuration, as well as a sample message from each sensor input. Ideally, if you could turn on debug mode and provide the output file up to and including the time the filter diverges, that would be great.

We use the sensor config to create a mask vector for the input source in question. Values that are not set to `true` will be masked out.

The only exception to this is when you have a transform defined for your sensor. The mask vectors are defined in sensor space, not EKF state space, so if you have values that, once transformed into the target frame, provide Z values to the filter, they will (and should) be used.

herolada, 2025-08-20:
Thank you, all clear! Here is the static `base_link->dgps_main_antenna`:
```
- Translation: [-0.090, 0.220, 0.130]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [0.000, -0.000, 0.000]
- Rotation: in RPY (degree) [0.000, -0.000, 0.000]
```

herolada, 2025-08-21:
Yes, running on the same bag file it looks good, no divergence, good covariance matrix. Seems fixed, thank you!

ayrton04, 2025-08-28:
Fixed in #942. Thanks!

closed 2025-08-28 by ayrton04